lkilcher/dolfyn #3: error in reading .vec files

fairybluebird, 2015-10-22:
Hello,

Firstly, thanks a ton for this incredibly useful dolfyn package! I am trying to implement your motion correction module to some ADV-IMU data that I have, and am running into issues. While I verified that the dolfyn package is correctly installed (it runs fine for the sample .vec files in the example_data folder), I get an error when I try to use it on the .vec files of my data. Does the package work for data acquired in both continuous as well as burst modes in the Vector?

[image]

I have attached a snapshot of the error here and also attached the sample acquired data in .vec format. Any comments regarding this would be extremely helpful.
https://www.dropbox.com/s/yr76kx32zfpuhgl/sample_data1.vec?dl=0

Thanks!
Aditya

lkilcher, 2015-10-23:
Thanks for identifying this bug, providing example data, and for your interest in dolfyn. I made some fixes that I believe resolve this issue. Was that data collected in burst mode? It looks like continuous data to me. Currently dolfyn isn't setup well to handle burst-mode data, but it is something that I've added to the TODO.txt. If you have an example file that uses burst mode duty-cycling, I'd be happy to make sure it loads correctly.

closed 2015-10-23 by lkilcher

fairybluebird, 2015-10-28:
Thanks for the prompt response and resolving the issue! I believe that the motion correction works now with my data - I am just playing with the input parameters and figuring out the best ones for my data. You are right, the data was collected in continuous mode. Thanks again for your help - I will get back if there are further issues in using dolfyn.

fairybluebird, 2016-03-29:
Hello,

An update on the fix: it works well with 90% of the ADV files that I have been running the motion correction algorithm on (all of them continuous mode). However, I have attached a sample run which I tried running the script recently where it does not. I have verified that the .vec file is not corrupted by extracting the .sen and .dat files from the same .vec file in question using Vector software. Any help would be again very much appreciated, sorry for the bother!
[image]

Screenshot of error attached, and link to sample vec file below:
https://www.dropbox.com/s/aeht38jlqigm8jl/Station40_20150921121545.vec?dl=0

Thanks!
Aditya