lkilcher/dolfyn #6: error: unpack requires a string argument of length 2 when reading adcp file

rmduarte, 2016-12-02:
Dear Levi,

After correcting the problem with the nanmean import, we get the following error when trying to read a adcp binary file:

In [23]: data = apm.read_rdi('SMADCP-FEM-Brehat-1.000')
---------------------------------------------------------------------------
error                                     Traceback (most recent call last)
<ipython-input-23-be1d361f7c0a> in <module>()
----> 1 data = apm.read_rdi('SMADCP-FEM-Brehat-1.000')

C:\Données brutes\dolfyn\dolfyn\adp\api.py in read_rdi(fname, nens)
     14
     15 def read_rdi(fname, nens=None):
---> 16     with adcp_loader(fname) as ldr:
     17         dat = ldr.load_data(nens=nens)
     18     return dat

C:\Données brutes\dolfyn\dolfyn\adp\_readbin.py in __init__(self, fname, navg, a
vg_func)
    597         #self.f=io.npfile(fname,'r','l')
    598         self.f = bin_reader(fname)
--> 599         self.read_hdr()
    600         self.read_cfg()
    601         # Seek back to the beginning of the file:

C:\Données brutes\dolfyn\dolfyn\adp\_readbin.py in read_hdr(self)
    480         cfgid = list(fd.read_ui8(2))
    481         nread = 0
--> 482         while (cfgid[0] != 127 | cfgid[1] != 127) | (not self.checkheade
r()):
    483             nextbyte = fd.read_ui8(1)
    484             pos = fd.tell()

C:\Données brutes\dolfyn\dolfyn\adp\_readbin.py in checkheader(self)
    770         if numbytes > 0:
    771             fd.seek(numbytes - 2, 1)
--> 772             cfgid = fd.read_ui8(2)
    773             #### sloppy code:
    774             if len(cfgid) == 2:

C:\Données brutes\dolfyn\dolfyn\adp\_read_bin.py in read_ui8(self, n)
    164
    165     def read_ui8(self, n):
--> 166         return self.read(n, 'B')
    167
    168     def read_float(self, n):

C:\Données brutes\dolfyn\dolfyn\adp\_read_bin.py in read(self, n, frmt)
    161             return unpack(self.endian + frmt * n, val)[0]
    162         else:
--> 163             return np.array(unpack(self.endian + frmt * n, val))
    164
    165     def read_ui8(self, n):

error: unpack requires a string argument of length 2

Do you have any idea? We can send you a data file to test the code.

Best regards,

Rui Duarte

rmduarte, 2016-12-02:
We are using Anaconda 4.2.0 and windows 7.

Thanks

lkilcher, 2016-12-02:
Which Python version are you using? It looks like Anaconda 4.2.0 offers a choice between Python 3.5 and 2.7.

rmduarte, 2016-12-02:
Sorry, Python 2.7

lkilcher, 2016-12-02:
Well, this is very strange. I am not getting the error you have indicated above, but I am getting a different one. Regarding the error you've indicated above, is there any other text printed to the console prior to the traceback? If so, can you post that text as well? Are you sure the file you're using is the same one Jean-Francois sent me? The one I have is 2078507008 bytes long (~1.9GB).

Regarding the error I'm getting: this file has a different header than other RDI ADCPs I've typically worked with. Can you let me know the model name/number of the instrument, and any software you used to configure the instrument, read data from the instrument, or post-process the data?

lkilcher, 2016-12-03:
OK, I resolved this with commit e59e37f. It had to do with a difference in the way that Windows reads files compared to unix. I needed to add a little `'b'` to the `open()` call. I wonder how many people have tried to use this code to load files in windows, and haven't been able to? This took ENTIRELY TOO LONG to figure out! Though, I suppose I've mostly been trying to figure out the header-discrepancy issue, which is now covered in issue #7.

closed 2016-12-03 by lkilcher