lkilcher/dolfyn #7: Why do '7f79' bytes appear in an RDI file where '7f7f' is expected?

lkilcher, 2016-12-03:
A file that I received from some collaborators has a different header start byte than what is indicated in the RDI documentation. I still do not understand why this is, but branch [debug-7f79](https://github.com/lkilcher/dolfyn/tree/debug-7f79) resolves the issue for the file they supplied. For now I'm going to leave this fix in that branch until I understand the source of the discrepancy.

lkilcher, 2016-12-03:
@rmduarte : I think you should be able to load your files by switching to the debug-7f79 branch.

rmduarte, 2016-12-05:
Hi Levi. Thanks a lot for your support. We were able to read the data file.
I don't have the details about the model used, but Jeff will contact you later with that information.

mmartini-usgs, 2017-04-06:
797F (depending on your endianness) is the ID for waves packets.

mmartini-usgs, 2017-04-06:
You can split the data with this code:
http://trac.nccoos.org/dataproc/browser/DPWP/trunk/DPWP/ADCP_splitter/pd0.py