lkilcher/dolfyn #14: Issue with read_nortek function

Tkideau, 2017-07-06:
Hello lklicher,
I was planning to develop a Python script to read .vec file when I found your library : very useful ! Many thanks for developing it ! 
I began to use your _read_nortek_ function and it works well on the majority of my .vec file but on some others, it throws this error :
[image]
I saw a similar issue was raised in the past and partially solved : https://github.com/lkilcher/dolfyn/issues/3
Unfortunately I can't provide you the .vec that are raising the error but the thing I can tell you is that data are generated in continuous mode.

Tkideau, 2017-07-19:
Hello lklicher,
I noted another issue perhaps related to this one : 
when processing some .vec files, the **while loop** in the _findnextid()_ method (<dolfyn.io.nortek>line 966) turns into an infinite loop. Variable **nowid** alternate between values 0 and 165 but never go to 17 (which seems to be the typical termination condition).
I observed the same in the case of "Reached the end of the file." issue. I thus thinking that the problem is around here (**searchcode**, _read_id()_, _read()_ ...).
Hope you will find a bit of time to look at this !
Regards,
Tkideau

lkilcher, 2017-07-25:
Hi and sorry for the slow response. It's extremely hard for me to debug these kinds of errors without the source file. If there is any way to share the data file with me I can assure you I won't share it with anyone else or use it for anything other than debugging. Otherwise, it's really not worth my time to try to guess what the problem is.

lkilcher, 2019-02-27:
Closing this due to lack of: activity and data.

closed 2019-02-27 by lkilcher