lkilcher/dolfyn #21: New Pip Install?

galenegan, 2018-06-18:
Hello,

First, many thanks for writing this package. It's going to help immensely with my data processing. However I'm having some trouble with the install. The most recent pip install is version 0.8.1, which encounters various errors in Python 3. And when I clone 0.9.0 from git, I can't get past the:

python setup.py install

because I encounter the error:

error: Unexpected HTML page found at https://github.com/lkilcher/pyDictH5.git#egg=pyDictH5

Anyway, I was hoping that by updating the pip version then the process would go more smoothly. Or if you have any other suggestions regarding the HTML egg error I'd be happy to try those as well!

-Galen

lkilcher, 2018-11-12:
Sorry that I've been so slow to respond. Did you get this working? If not, you should try cloning and installing the latest version of [pyDictH5](http://github.com/lkilcher/pyDictH5)?

lkilcher, 2018-11-13:
I believe commit 94a80c0 resolves this.