lkilcher/dolfyn #43: Declination adjustment

lkilcher, 2019-03-25:
Does the sign of the declination adjustment change if an ADP or ADV is up vs. down-facing?

What about the order of the rotations? Does it matter?

mmartini-usgs, 2019-03-26:
Yes, the order matters.
I would make the declination adjustment to the heading value.  Then apply the rotations.

I have had considerable trouble with ADV rotations for cabled Vectors.  Are you having issues with this?  I have a very good test data set.

mmartini-usgs, 2019-03-26:
Here is a little more from my notes:
Nortek documentation indicates that +z is always into the probe. However, this can vary depending
on how the firmware interprets the up/down bit and other unknown factors. To know for sure the direction of +z, one can examine the last row of the transformation matrix provided in the .hdr or .whr files. This is the matrix T in Nortek’s transformation code for MATLAB.  If the third row of T is negative, +z is out of the probe.  Therefore: If the Nortek supplied transformation matrix row 3 is negative, do nothing, +z is out of the probe. To rotate it back into the probe, rotate about the x axes by applying the transformation matrix:
```
     | 1 0 0 |
 A = | 0 -1 0 |
     | 0 0 -1 |
```

lkilcher, 2019-03-26:
I'm certainly interested in your test data set. What does it look like?

I think what you're describing is documented in the `refs/transform.m` script (which I got from the old Nortek forum website). Part of my concern here has to do with the question: how does the *orientation matrix* from an IMU (when the ADV has one) compare with the *rotation matrix* described in that script? Previously I'd thought I was tackling this correctly, but now I'm realizing there *might* be a bug here. So, this requires a bit more attention.

lkilcher, 2019-03-26:
Also note that this is all very closely related to #44.

lkilcher, 2019-03-26:
Also, regarding your comment:

> I have had considerable trouble with ADV rotations for cabled Vectors. Are you having issues with this?

I think this is one thing that DOLfYN does fairly well. The tricky part is knowing what the rotation matrix between the body-frame and head-frame is. If you get that right, and put it into `dat.props['body2head_rotmat']`, and also define `dat.props['body2head_vec']` correctly, then using `dolfyn.adv.motion.CorrectMotion` should do the rest.

mmartini-usgs, 2019-03-26:
Hi Levi,

Hmm.  Attached is what I worked out for that experiment.  Nortek is using
“mixed” conventions, as far as I can tell, and that really tripped things
up.  I wrote the tome to keep track of what I was doing and why in my
code.  It was not a simple problem and Nortek was not particularly helpful.

This is the data –

https://stellwagen.er.usgs.gov/mvco_14.html

https://stellwagen.er.usgs.gov/mvco_15.html

I can send you the raw stuff to work with if you like.

This is what the platforms looked like:

https://soundwaves.usgs.gov/2014/10/fieldwork2.html

https://soundwaves.usgs.gov/2017/02/research4.html

Our collaborators at WHOI threw up their hands and basically rotate their
data to match the principle axis of the current and call it a day.  OK if
you are looking for turbulence, lousy if you want to actually measure the
current with the ADV, so I tried to solve the problem.

Best regards,

Marinna

*From:* Levi Kilcher <notifications@github.com>
*Sent:* Tuesday, March 26, 2019 2:18 PM
*To:* lkilcher/dolfyn <dolfyn@noreply.github.com>
*Cc:* Marinna Martini <mmartini@usgs.gov>; Comment <
comment@noreply.github.com>
*Subject:* [EXTERNAL] Re: [lkilcher/dolfyn] Declination adjustment (#43)

Also, regarding your comment:

I have had considerable trouble with ADV rotations for cabled Vectors. Are
you having issues with this?

I think this is one thing that DOLfYN does fairly well. The tricky part is
knowing what the rotation matrix between the body-frame and head-frame is.
If you get that right, and put it into dat.props['body2head_rotmat'], and
also define dat.props['body2head_vec'] correctly, then using
dolfyn.adv.motion.CorrectMotion should do the rest.

—
<https://github.com/lkilcher/dolfyn/issues/43#issuecomment-476782775>, or mute
the thread
.[image: Image removed by sender.]

lkilcher, 2019-05-24:
I believe this was resolved in #50. Note that this added a new method, `dat.set_declination`, which should be used to set the declination.

closed 2019-05-24 by lkilcher