lkilcher/dolfyn #61: edit suggested for assigning dat['range'] for AD2CP data

mcfogarty, 2019-06-24:
in nortek2.py
data['range'] = (np.arange(data['vel'].shape[1]) *
data['config']['cell_size'] +
data['config']['blanking'])
if 'vel_b5' in data:
data['range_b5'] = (np.arange(data['vel_b5'].shape[1]) *
data['config']['cell_size_b5'] +
data['config']['blanking_b5'])
 
 
But, np.arange(data['vel'].shape[1]) yields an
array with first entry as 0
 
if cell_size = 1.0 m and blanking = 0.5 m,
dat.range[0] = 0 * cell_size + blanking = 0 + 0.5 = 0.5, when based on 
N3015-025-Principles-of-Operation_Signature document from Nortek,
 
dat.range[0] should be 
 
center of nth cell = blanking + n * cell_size = 0.5 m + (1)*1 m = 1.5 m
 
Suggested edit to nortek2.py:
data['range'] = (np.arange(data['vel'].shape[1]) + 1 *
data['config']['cell_size'] +
data['config']['blanking'])
if 'vel_b5' in data:
data['range_b5'] = (np.arange(data['vel_b5'].shape[1]) + 1 *
data['config']['cell_size_b5'] +
data['config']['blanking_b5'])
 
Note the rdi.py read file assigns dat['range'] differently and is okay.
dat['range'] = (self.cfg['bin1_dist_m'] +
np.arange(self.cfg['n_cells']) *
self.cfg['cell_size_m'])

lkilcher, 2021-12-24:
@jmcvey3 I beleive you've resolved this in #67, right?

jmcvey3, 2021-12-26:
> @jmcvey3 I beleive you've resolved this in #67, right?

That is correct.

lkilcher, 2021-12-27:
Closing since this will be resolved with #67.

closed 2021-12-27 by lkilcher