lkilcher/dolfyn #62: Error in io._read_bin

leewujung, 2019-08-06:
I ran into the following error when trying to read source data files in example_data:
```
In[16]: dolfyn.read('RDI_withBT.000')
Reading file RDI_withBT.000 ...
Traceback (most recent call last):
  File "/Users/wu-jung/miniconda3/envs/echopype/lib/python3.7/site-packages/IPython/core/interactiveshell.py", line 3291, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-16-1b2a3777040f>", line 1, in <module>
    dolfyn.read('RDI_withBT.000')
  File "/Users/wu-jung/miniconda3/envs/echopype/lib/python3.7/site-packages/dolfyn/io/api.py", line 35, in read
    dat = func(fname, userdata=userdata, nens=nens)
  File "/Users/wu-jung/miniconda3/envs/echopype/lib/python3.7/site-packages/dolfyn/io/rdi.py", line 38, in read_rdi
    with adcp_loader(fname) as ldr:
  File "/Users/wu-jung/miniconda3/envs/echopype/lib/python3.7/site-packages/dolfyn/io/rdi.py", line 831, in __init__
    self.read_hdr()
  File "/Users/wu-jung/miniconda3/envs/echopype/lib/python3.7/site-packages/dolfyn/io/rdi.py", line 699, in read_hdr
    nextbyte = fd.read_ui8(1)
  File "/Users/wu-jung/miniconda3/envs/echopype/lib/python3.7/site-packages/dolfyn/io/_read_bin.py", line 171, in read_ui8
    return self.read(n, 'B')
  File "/Users/wu-jung/miniconda3/envs/echopype/lib/python3.7/site-packages/dolfyn/io/_read_bin.py", line 163, in read
    raise eofException
dolfyn.io._read_bin.eofException
```
I tested it using python 3.7 and 3.5. The same error occurred for reading ad2cp files and a few others I tried to read.
This must be something stupid but I couldn't figure it out. Any help will be much appreciated!

lkilcher, 2019-08-06:
What system are you on (win, mac, linux)? Can you share a data file that is
generating the error message?

leewujung, 2019-08-06:
I am on a Mac. The data file I tried to read is one of the example files: [dolfyn/dolfyn/example_data/RDI_withBT.000](https://github.com/lkilcher/dolfyn/blob/master/dolfyn/example_data/RDI_withBT.000)

Thanks!

lkilcher, 2019-08-06:
Did you git-clone the repo, or install via pip? Assuming the former, this
is probably because you don't have git-lfs installed. Go here
<https://dolfyn.readthedocs.io/en/stable/install.html#data-files-and-test-files>,
and especially here
<https://help.github.com/en/articles/installing-git-large-file-storage> for
more info.

If you installed via pip this may be a bug in the pip distribution.

leewujung, 2019-08-06:
You are right that I messed up the git lfs part. I installed git-lfs using brew and executed `git lfs install` once earlier. For some reason when I do `git lfs install` again just now, it updated the hook and was able to pull the files correctly. 

The problem is now resolved. Thanks for responding so promptly!

closed 2019-08-06 by leewujung