lkilcher/dolfyn #69: Deprecate `do_avg`, `do_var` and `do_tke`?

lkilcher, 2021-12-24:
Not sure these functions are that valuable, and they add bulk to the codebase. I suggest we drop them. I still think the `__call__` method (which just averages everything) is useful, which calls `do_avg`, so we'll need to move that functionality somewhere. Maybe for now we just move it to `_do_avg`?

@jmcvey3, do you like them? If you don't have a strong opinion, I think we should drop those before merging #67.

jmcvey3, 2021-12-26:
`do_avg` is a core function for ensembling the dataset, so how would you propose doing that if deprecated?
I agree that `do_tke` adds bulk.

lkilcher, 2022-01-14:
Roger. Since `do_avg` is a major piece of the API, let's just leave `do_tke` in for now.

lkilcher, 2022-01-14:
Also, I've changed my mind on the `__call__` method. Let's leave it out in favor of the more explicit `do_avg`.

closed 2022-01-14 by lkilcher