lkilcher/dolfyn #83: Problème de lecture de fichiers ntk sous python

bfanch, 2022-01-31:
Bonjour, 
J'essaie de charger un fichier ntk sous python avec la librairie dolfyn mais la lecture plante,
Je ne comprends pas où est le problème. J'ai essayé d'autres fichiers ntk mais cela fini toujours pareil. Il boucle en cherchant une valeur valide de cfgid. Je ne comprends pas l'erreur, est-ce que vous pourriez m'aider?
Je joins le fichier que j'essaie de lire (c'est un fichier enregistré dans une poubelle remplie d'eau, les données sont forcément très mauvaises...

https://wetransfer.com/downloads/72a01009e17cc8f971e4611059e61c0920220131101040/7194a7c81869dd7d8c7f5a0126d7ad1420220131101058/155e59

merci beaucoup,
François

In [28]: f3=dolfyn.read(file3)

Reading file SIG500_SW.208.21.SIG500_SW.00000.ntk ...
  Still looking for valid cfgid at file position 1000 ...
  Still looking for valid cfgid at file position 2000 ...
  Still looking for valid cfgid at file position 3000 ...
....
Still looking for valid cfgid at file position 97000 ...
  Still looking for valid cfgid at file position 98000 ...
  Still looking for valid cfgid at file position 99000 ...
Traceback (most recent call last):

  File "<ipython-input-19-795f1adbb195>", line 1, in <module>
    f1=dolfyn.read(file1)

  File "c:\scripts\lecture_adcp\dolfyn-master\dolfyn-master\dolfyn\io\api.py", line 35, in read
    dat = func(fname, userdata=userdata, nens=nens)

  File "c:\scripts\lecture_adcp\dolfyn-master\dolfyn-master\dolfyn\io\rdi.py", line 38, in read_rdi
    with adcp_loader(fname) as ldr:

  File "c:\scripts\lecture_adcp\dolfyn-master\dolfyn-master\dolfyn\io\rdi.py", line 834, in __init__
    self.read_hdr()

  File "c:\scripts\lecture_adcp\dolfyn-master\dolfyn-master\dolfyn\io\rdi.py", line 702, in read_hdr
    nextbyte = fd.read_ui8(1)

  File "c:\scripts\lecture_adcp\dolfyn-master\dolfyn-master\dolfyn\io\_read_bin.py", line 171, in read_ui8
    return self.read(n, 'B')

  File "c:\scripts\lecture_adcp\dolfyn-master\dolfyn-master\dolfyn\io\_read_bin.py", line 163, in read
    raise eofException

eofException

lkilcher, 2022-02-01:
Hi @bfanch, DOLfYN doesn't currently support reading `.ntk` files. From your email it sounds like you're trying to read these files in realtime? This adds to the complexity, and I'm not sure how much help DOLfYN can be for this. It's something we might be interested in implementing at some point, but I don't have time to tackle it now. Also, I'm not sure that the `.ntk` format is published, which also adds to the complexity. I wonder if @jmcvey3 has any insight here?

jmcvey3, 2022-02-01:
@bfanch I'm assuming you are using the Nortek Midas software? Is it possible to accomplish what you are doing in Nortek's Signature Deployment software? It can save ADCP data live into `.ad2cp` files, which can be subsequently read by dolfyn.