lkilcher/dolfyn #96: Amplitude and correlation coordinates changing and possibly incorrect

dnowacki-usgs, 2022-03-29:
For Nortek Vector instruments, amplitude and correlation are quantities related to the beams and do not depend on the measurement coordinate system. In DOLfYN, however, the `dir` coordinate of `amp` and `corr` is that of the deployment, instead of beam. Additionally, since these variables use the `dir` coordinate, the coordinates change after rotating.  The values themselves do not change (since they are not in the `rotate_vars` attr), but their coordinates do, which is unexpected and I believe incorrect behavior. 

I think the solution here is always to map `amp` and `corr` to beam coordinates and not the `dir` coordinate. I'm not yet familiar enough with the code to propose a PR, but happy to help if I can.

jmcvey3, 2022-04-13:
Good catch! Thanks for bringing this up. I'll add this as a hotfix to the latest release once completed.

jmcvey3, 2022-07-07:
Closing as this particular issue is fixed.