lkilcher/dolfyn #98: Bin mapping

jessecusack, 2022-04-21:
Thank you for developing a nice package! I am trying to process a TRDI workhorse that pitched 10 - 15 degrees and needs bin mapping prior to coordinate rotations. Correct me if I am wrong, but it doesn't seem like you have bin mapping functionality in here? If I were to attempt to add this, where would be the appropriate place?

jmcvey3, 2022-04-22:
@jessecusack Thank you! You are correct, we have not tried to tackle bin mapping. I appreciate you wanting to attempt this; it would fall under the 'ADP' api, so would be added as a new file in the dolfyn/adp directory.

apaloczy, 2022-08-12:
Just found this issue. Thank you for developing DOLfYN!

@jessecusack, @jmcvey3, I started implementing bin-mapping following [Dewey & Stringer (2007)](http://dx.doi.org/10.13140/RG.2.1.1042.8002)'s Equation B1 [here](https://github.com/apaloczy/dewaveADCP/blob/894c836ecf1e248135c17fdcae418d5ee238d0cd/dewaveADCP/beam2earth.py#L430) a while ago. It could use more testing, but some sanity checks with simple synthetic velocity fields and different pitch/roll angles seem to give qualitatively correct results. Do you think this function might be useful as a starting point?

lkilcher, 2022-12-21:
Let's get a PR for this ASAP!

jessecusack, 2023-01-05:
I ended up implementing bin mapping in a different package, the code is here:

 https://github.com/modscripps/velosearaptor/blob/9c70a0b9ab376ca14a416d4e5fae07340b65a0b0/velosearaptor/madcp.py#L897
 
One advantage of the linear interpolation method I use in the above is that it is vectorized and therefore quite fast. A downside is that it creates a rather large arrays that hog RAM. 

I guess parts of this or @apaloczy's package could be copied into dolfyn relatively easily, but I don't have time to figure it myself.

jmcvey3, 2023-01-05:
Thanks @jessecusack. The MODScripps code is implemented in TRDI's coordinate system, correct? 

The copying in should be easy, yes; testing and making sure the coordinate systems are correct would be the hard part.

jessecusack, 2023-01-05:
Yes it does use the TRDI coordinate system.