lkilcher/dolfyn #101: Incorrect version numbering on conda?

jessecusack, 2022-06-29:
When I install dolfyn through `conda-forge` and I got version 0.13. I feel like the actual version number should be 1.0, since my data is loaded in an xarray object and the README indicates this is a 1.0 feature. Not sure what is going on there...

jmcvey3, 2022-12-19:
@lkilcher Any ideas on what pipeline is uploading dolfyn to conda-forge? (https://anaconda.org/conda-forge/dolfyn/files). Appears to auto-run whenever I push to pypi, and I'm assuming conda-forge is still running on the 0.13 recipe for dependencies and whatnot.

lkilcher, 2023-01-03:
I just did `conda install -c conda-forge dolfyn`, and everything worked seemlessly. Also, the version number throughout the install process and after install always says "1.2.0". I suspect this is from the time period when we were preparing to release 1.0 (the pre-release was 0.13).

closed 2023-01-03 by lkilcher