lkilcher/dolfyn #108: Waves parameters from the 5th beam

bfanch, 2022-09-12:
Hi,
I'm trying to calculate the wave parameters from the 5th beam of a Nortek ADCP. 
I was wondering(/hoping) if there were something already coded in the library dolfyn?
Thanks for your help,
François

lkilcher, 2022-09-12:
François-

Thanks for your interest in DOLfYN. Unfortunately this feature is not implemented, or planned to be. But, if you develop these tools, it sounds like something that would be a valuable addition to the package.

jmcvey3, 2023-01-20:
Just occurred to me: if "from the 5th beam of a Nortek ADCP", you mean the acoustic surface tracking (AST) data, dolfyn can read that in. You can use `calc_psd` to calculate the vertical wave spectra, and then import [MHKiT](https://github.com/MHKiT-Software/MHKiT-Python/blob/master/mhkit/wave/resource.py) to calculate the wave stats.