lkilcher/dolfyn #114: Many standard_name values assigned to variables are not actually standard_names

dnowacki-usgs, 2023-03-21:
I'm super happy that DOLfYN includes support for CF Conventions (see #102, #113). In processing some Nortek Signature data and running it through the [IOOS Compliance Checker](https://github.com/ioos/compliance-checker), I see that many variables have `standard_name` values that are not in the [CF Standard Name Table](https://cfconventions.org/Data/cf-standard-names/current/build/cf-standard-name-table.html).

My understand is if a `standard_name` does not exist for a given variable, short of being proposed as a new `standard_name`, it should be listed as a `long_name` attr instead. 

**Output from compliance checker:**
```
--------------------------------------------------------------------------------
                         IOOS Compliance Checker Report
                                 Version 5.0.2
                     Report generated 2023-03-21T22:16:57Z
                                     cf:1.8
http://cfconventions.org/Data/cf-conventions/cf-conventions-1.8/cf-conventions.html
--------------------------------------------------------------------------------
                               Corrective Actions
dolfyntest.nc has 5 potential issues

                                     Errors
--------------------------------------------------------------------------------
§2.2 Data Types
* The variable beam failed because the datatype is int64
* The variable dir failed because the datatype is int64
* The variable x failed because the datatype is int64
* The variable x* failed because the datatype is int64

§3.1 Units
* units for xmit_energy, "dB" are not recognized by UDUNITS
* units for echo, "dB" are not recognized by UDUNITS
* units for xmit_energy_echo, "dB" are not recognized by UDUNITS
* units for amp, "dB" are not recognized by UDUNITS
* units for xmit_energy_b5, "dB" are not recognized by UDUNITS
* units for amp_b5, "dB" are not recognized by UDUNITS

§3.3 Standard Name
* Attribute long_name or/and standard_name is highly recommended for variable boost_running
* Attribute long_name or/and standard_name is highly recommended for variable low_volt_skip
* Attribute long_name or/and standard_name is highly recommended for variable active_config
* Attribute long_name or/and standard_name is highly recommended for variable telemetry_data
* Attribute long_name or/and standard_name is highly recommended for variable x
* Attribute long_name or/and standard_name is highly recommended for variable x*
* standard_name velocity_from_multibeam_acoustic_doppler_velocity_profiler_in_sea_water is not defined in Standard Name Table v80. Possible close match(es): ['indicative_error_from_multibeam_acoustic_doppler_velocity_profiler_in_sea_water', 'beam_consistency_indicator_from_multibeam_acoustic_doppler_velocity_profiler_in_sea_water', 'signal_intensity_from_multibeam_acoustic_doppler_velocity_sensor_in_sea_water']
* standard_name undefined is not defined in Standard Name Table v80.
* standard_name platform_angular_velocity is not defined in Standard Name Table v80. Possible close match(es): ['upward_air_velocity', 'platform_yaw_angle', 'platform_view_angle']
* standard_name undefined is not defined in Standard Name Table v80.
* standard_name sound_pressure_level_of_acoustic_signal is not defined in Standard Name Table v80. Possible close match(es): ['sound_pressure_level_in_air', 'sound_pressure_level_in_water', 'sound_pressure_in_air']
* standard_name platform_acceleration is not defined in Standard Name Table v80. Possible close match(es): ['platform_orientation', 'platform_heave_rate_down', 'platform_heave_rate']
* standard_name platform_acceleration is not defined in Standard Name Table v80. Possible close match(es): ['platform_orientation', 'platform_heave_rate_down', 'platform_heave_rate']
* standard_name sound_pressure_level_of_acoustic_signal is not defined in Standard Name Table v80. Possible close match(es): ['sound_pressure_level_in_air', 'sound_pressure_level_in_water', 'sound_pressure_in_air']
* standard_name battery_voltage is not defined in Standard Name Table v80. Possible close match(es): ['altimeter_range']
* standard_name echo_sounder_signal_intensity is not defined in Standard Name Table v80.
* standard_name magnetic_field_vector is not defined in Standard Name Table v80.
* standard_name signal_intensity_from_multibeam_acoustic_doppler_velocity_profiler_in_sea_water is not defined in Standard Name Table v80. Possible close match(es): ['signal_intensity_from_multibeam_acoustic_doppler_velocity_sensor_in_sea_water', 'indicative_error_from_multibeam_acoustic_doppler_velocity_profiler_in_sea_water', 'beam_consistency_indicator_from_multibeam_acoustic_doppler_velocity_profiler_in_sea_water']
* standard_name platform_angular_velocity is not defined in Standard Name Table v80. Possible close match(es): ['upward_air_velocity', 'platform_yaw_angle', 'platform_view_angle']
* standard_name sound_pressure_level_of_acoustic_signal is not defined in Standard Name Table v80. Possible close match(es): ['sound_pressure_level_in_air', 'sound_pressure_level_in_water', 'sound_pressure_in_air']
* standard_name magnetic_field_vector is not defined in Standard Name Table v80.
* standard_name platform_acceleration is not defined in Standard Name Table v80. Possible close match(es): ['platform_orientation', 'platform_heave_rate_down', 'platform_heave_rate']
* standard_name platform_quaternion_vector is not defined in Standard Name Table v80. Possible close match(es): ['platform_surge_rate_fore', 'platform_surge_fore', 'platform_sway_rate_port']
* standard_name undefined is not defined in Standard Name Table v80.
* standard_name magnetic_field_vector is not defined in Standard Name Table v80.
* standard_name platform_angular_velocity is not defined in Standard Name Table v80. Possible close match(es): ['upward_air_velocity', 'platform_yaw_angle', 'platform_view_angle']
* standard_name undefined is not defined in Standard Name Table v80.
* standard_name platform_quaternion_vector is not defined in Standard Name Table v80. Possible close match(es): ['platform_surge_rate_fore', 'platform_surge_fore', 'platform_sway_rate_port']
* standard_name undefined is not defined in Standard Name Table v80.
* standard_name undefined is not defined in Standard Name Table v80.
* standard_name undefined is not defined in Standard Name Table v80.
* standard_name undefined is not defined in Standard Name Table v80.
* standard_name signal_intensity_from_multibeam_acoustic_doppler_velocity_profiler_in_sea_water is not defined in Standard Name Table v80. Possible close match(es): ['signal_intensity_from_multibeam_acoustic_doppler_velocity_sensor_in_sea_water', 'indicative_error_from_multibeam_acoustic_doppler_velocity_profiler_in_sea_water', 'beam_consistency_indicator_from_multibeam_acoustic_doppler_velocity_profiler_in_sea_water']
* standard_name battery_voltage is not defined in Standard Name Table v80. Possible close match(es): ['altimeter_range']
* standard_name velocity_from_multibeam_acoustic_doppler_velocity_profiler_in_sea_water is not defined in Standard Name Table v80. Possible close match(es): ['indicative_error_from_multibeam_acoustic_doppler_velocity_profiler_in_sea_water', 'beam_consistency_indicator_from_multibeam_acoustic_doppler_velocity_profiler_in_sea_water', 'signal_intensity_from_multibeam_acoustic_doppler_velocity_sensor_in_sea_water']
* standard_name undefined is not defined in Standard Name Table v80.
* standard_name undefined is not defined in Standard Name Table v80.
* standard_name platform_quaternion_vector is not defined in Standard Name Table v80. Possible close match(es): ['platform_surge_rate_fore', 'platform_surge_fore', 'platform_sway_rate_port']

                                    Warnings
--------------------------------------------------------------------------------
§2.3 Naming Conventions
* dimension x* should begin with a latter and be composed of letters, digits, and underscores
* variable x* should begin with a letter and be composed of letters, digits, and underscores

§2.6 Attributes
* §2.6.2 global attribute title should exist and be a non-empty string
* §2.6.2 global attribute history should exist and be a non-empty string
* §2.6.1 Conventions field is not present
```

jmcvey3, 2023-05-23:
Hi @dnowacki-usgs, thanks for linking the IOOS checker, that's a useful tool. I'll look into that and update the relevant names.

We don't plan on breaking down velocity into its directional components, which have standard names, but would appreciate thoughts on handling metadata on that.

Also, if there is a process to add new standard names, there's no differentiation between ADCP and ADV 
"acoustic signal amplitude" (beam_consistency_indicator_from_multibeam_acoustic_doppler_velocity_profiler_in_sea_water) and "acoustic signal correlation" (signal_intensity_from_multibeam_acoustic_doppler_velocity_sensor_in_sea_water),
and that would be good to update.

jmcvey3, 2023-11-07:
Closing with PR merge

closed 2023-11-07 by jmcvey3