lkilcher/dolfyn #130: 'roll' variable is not being created in xarray dataset

BecCowley, 2024-06-29:
I'm using the dolfyn package to read RDI LADCP data and have run into an interesting issue. I am not sure if this is my python set up, so putting it here for someone else to test and advise on.

When converting the ds_dict (line 251 of io/base.py, the _create_dataset function) to a xarray dataset, the 'roll' variable is not converted. However, a 'roll()' method is available in the dataset.
Xarray documentation on the 'roll' method is referenced here: https://docs.xarray.dev/en/stable/generated/xarray.IndexVariable.html#xarray.IndexVariable

If I limit the conversion to just the 'roll' variable and not all the others, the conversion works.

I can't work out why the roll is not converted from the full ds_dict. Maybe xarray is preferencing the 'roll' method?
I'm using Xarray 2024.6.0 and I've installed Dolfyn as an editable package. Any ideas are welcome.
Thanks for a great package.

jmcvey3, 2024-07-02:
Hi @BecCowley, yes, xarray prefers the roll method if you try to call the variable via attribute. Does `dataset['roll']` not return the roll measurements?

BecCowley, 2024-07-02:
@jmcvey3 thanks, that works. I see now what is happening! I'll have to be more rigorous in my coding.

closed 2024-07-03 by jmcvey3