lkilcher/dolfyn #132: Error while reading .ad2cp file: the index file cannot be created

tdoda, 2025-04-09:
Hello,

I am trying to import a .ad2cp file from a Nortek Signature 1000 with dolfyn. It worked for past deployments for which we used the "burst using 5 beams" mode, but not for our last deployment, for which we used the "burst with echosounder" mode.   
I get the following error when trying to read the file with dolfyn.read:

"File ~\AppData\Local\anaconda3\envs\datalakes\Lib\site-packages\dolfyn\io\api.py:103 in read
    return func(fname, userdata=userdata, nens=nens, **kwargs)

  File ~\AppData\Local\anaconda3\envs\datalakes\Lib\site-packages\dolfyn\io\nortek2.py:66 in read_signature
    rdr = _Ad2cpReader(filename, rebuild_index=rebuild_index, debug=debug)

  File ~\AppData\Local\anaconda3\envs\datalakes\Lib\site-packages\dolfyn\io\nortek2.py:126 in __init__
    self._index = lib.get_index(fname,

  File ~\AppData\Local\anaconda3\envs\datalakes\Lib\site-packages\dolfyn\io\nortek2_lib.py:223 in get_index
    _create_index(infile, index_file, 2 ** 32, debug)

  File ~\AppData\Local\anaconda3\envs\datalakes\Lib\site-packages\dolfyn\io\nortek2_lib.py:147 in _create_index
    fin.seek(dat[4], 1)

OSError: [Errno 22] Invalid argument"

The error seems to come from the fact that dat[4] is a negative number. Here are the dat values when the error comes: (0, 192, 1, 0, -32768, -2, -24321). I am wondering if the ad2cp got corrupted while exporting it with the Nortek Signature Deployment software? I have tried with different data files, but the issue is the same. Note that all these .ad2cp files are correctly converted to .mat files with the deployment software. 
[Here is an example of an .ad2cp file that cannot be read.](https://github.com/user-attachments/files/19664736/S101024A024_float1.zip)

Thank you in advance for your help!

Best,

Tomy

jmcvey3, 2025-04-29:
Hi Tomy, thank you for bringing up this error - for your fyi, dolfyn was migrated over to mhkit (https://github.com/MHKiT-Software/MHKiT-Python), where it has continued to receive updates. 

I'm still getting an error the most recent version, so if you could copy your error over to that repository, I'd appreciate it.

jmcvey3, 2025-05-08:
@tdoda You can find the solution to your issue in this PR: https://github.com/MHKiT-Software/MHKiT-Python/pull/397

Note two datasets will get returned here, one with the echo sounder data and the other with the velocity data. 

The reason this one errored out for you originally was because the water velocity wasn't collected in "burst" mode, but in an "averaging" mode with a different sampling rate than the echo sounder. This is a use case we haven't seen before.

tdoda, 2025-05-14:
Great, thanks a lot!